indrekluuk/LiveOV7670 #1: error

lostgab, 2016-12-19:
cchpQUlM.ltrans0.o:(.text.startup+0x86): undefined reference to `initializeScreenAndCamera()'

cchpQUlM.ltrans0.o:(.text.startup+0x8e): undefined reference to `processFrame()'

collect2.exe: error: ld returned 1 exit status

indrekluuk, 2016-12-19:
What exactly did you do? Did you use Arduino IDE or platformIO?

It works with my platformIO + CLion project and I tried it with Arduino IDE (downloded the zip file from github and renamed "src" folder to "LiveOV7670") and it compiled without any errors.

lostgab, 2016-12-20:
Hey Indrek,
i am using the arduino ide, i changed the src folder to LiveOV7670 and it
says setup.h is missing
the i tried to copy the setup.h file in text form to another tab on arduino
and it did not work also,
what should i do now?
thanks
Sincerely Gabriel

indrekluuk, 2016-12-21:
It seems that you downloaded only the .ino file. You need all the files.

Download entire project:
[image]

After extracting the zip file you get folder "LiveOV7670-master". You need "LiveOV7670-master/arduino/src" folder. Rename the "src"->"LiveOV7670"

Additionally you need to copy "LiveOV7670-master/arduino/lib/Adafruit_GFX_Library"  to your arduino libraries folder (in you computer's home folder "Documents/Arduino/libraries") - If you have done anything with Adafruit graphics libraries before then you may already have this.

Then you should be able to open the .ino file and it should compile without errors

lostgab, 2016-12-22:
Hey Indrek,
i did as you instructed and i still got some weird error, can i ask for
help on what to do? thanks:)
Sincerely Gabriel
[image: Inline image 1][image: Inline image 2]
[image: Inline image 3]

indrekluuk, 2016-12-23:
What exactly the errors? I can't see any attachment images here. Could you upload the example images directly to here in github?

lostgab, 2016-12-23:
[image]

that is the image for my error

C:\Users\Gabri\AppData\Local\Temp\cccv5yFi.ltrans0.ltrans.o: In function `main':

cccv5yFi.ltrans0.o:(.text.startup+0x8a): undefined reference to `CameraOV7670::init()'

cccv5yFi.ltrans0.o:(.text.startup+0x94): undefined reference to `Adafruit_ST7735_mod::initR(unsigned char)'

cccv5yFi.ltrans0.o:(.text.startup+0xa0): undefined reference to `Adafruit_ST7735_mod::fillScreen(unsigned int)'

cccv5yFi.ltrans0.o:(.text.startup+0x14d8): undefined reference to `Adafruit_ST7735_mod::startAddrWindow(unsigned char, unsigned char, unsigned char, unsigned char)'

cccv5yFi.ltrans0.o:(.text.startup+0x1508): undefined reference to `Adafruit_ST7735_mod::endAddrWindow()'

C:\Users\Gabri\AppData\Local\Temp\cccv5yFi.ltrans3.ltrans.o: In function `global constructors keyed to 65535_0_ExampleTftBufferedCameraFrame.cpp.o.2004':

cccv5yFi.ltrans3.o:(.text.startup+0x3c): undefined reference to `CameraOV7670Registers::CameraOV7670Registers(unsigned char)'

cccv5yFi.ltrans3.o:(.text.startup+0x52): undefined reference to `Adafruit_ST7735_mod::Adafruit_ST7735_mod(signed char, signed char, signed char)'

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Arduino/Genuino Uno.

this is the error message, please help? thanks:)

indrekluuk, 2016-12-29:
I upgraded my Arduino IDE from 1.6 to 1.8 and got a similar error.
I will fix it this weekend and report back to here when you should try again.

indrekluuk, 2016-12-29:
It seems that Arduino IDE does not support cpp files in sub-folders anymore. Version 1.6 worked fine.
I will think of something I probably have to move them to the main folder to make it Arduino IDE compatible.

lostgab, 2016-12-30:
Thanks bro, looking forward to the update:)

indrekluuk, 2017-01-01:
I moved the the camera code to a library. Please try downloading the project again. But now you also have to copy LiveOV7670Library from the "lib" folder to your Arduino "libraries" folder. I tried with Arduino IDE 1.8 and it worked

lostgab, 2017-01-02:
nope still doesnt work bro

indrekluuk, 2017-01-02:
What is the error message? Did you copy "LiveOV7670Library" to the "libraries" folder?

lostgab, 2017-01-02:
C:\Users\Gabri\Downloads\LiveOV7670-master\LiveOV7670-master\arduino\src\LiveOV7670\LiveOV7670.ino:10:19:
fatal error: setup.h: No such file or directory

 #include "setup.h"

                   ^

compilation terminated.

exit status 1
Error compiling for board Arduino/Genuino Uno.

yes i did copy the library to the libraries folder
what should i do now thanks

indrekluuk, 2017-01-02:
It seems that you opened the .ino with Arduino IDE and it automatically created sub-folder named "LiveOV7670". The problem is that all the other files are still inside in "src" folder and Arduino IDE does not see them.  
  
You have to rename the "src" folder to "LiveOV7670" (or move all the fils from "src" to the automatically created "LiveOV7670" folder)

lostgab, 2017-01-02:
hey it solved the setup.h problem, how did u know? now there is another
problem,

C:\Users\Gabri\AppData\Local\Temp\ccHJPNuk.ltrans0.ltrans.o: In function
`setUpCamera':

C:\Users\Gabri\Documents\Arduino\libraries\LiveOV7670Library/CameraOV7670.cpp:37:
undefined reference to `CameraOV7670Registers::regsVGA'

C:\Users\Gabri\Documents\Arduino\libraries\LiveOV7670Library/CameraOV7670.cpp:37:
undefined reference to `CameraOV7670Registers::regsVGA'

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Arduino/Genuino Uno.

indrekluuk, 2017-01-02:
This is strange. It seems that Arduino IDE can't find cpp files in the library folder.

Does the content of the "LiveOV7670Library" look like this?
[image]

Did you re-download the whole project? Project folder should now look like this:

[image]

lostgab, 2017-01-03:
[image: Inline image 1]
[image: Inline image 2]

this is what i got

leonhart88, 2017-02-27:
i found this error on my 1.8 TFT LCD
how to fix it?
please
[image]

lostgab, 2017-02-27:
the connections must be right, i cant remember exactly, here is a youtube
video to help.https://youtu.be/A9EwJ7M7OsI

indrekluuk, 2017-03-01:
Does the example project included with the Adafruit library work?

This video:
https://www.youtube.com/watch?v=boagCpb6DgY&t=150s

lostgab, 2017-03-02:
yes:)

indrekluuk, 2017-03-02:
I updated to project so that at the beginning it will either flash entire screen to green if i2c connection to camera was successful or red if it failed.  
  
Could you please re-download the project and check if you see green or red screen after start-up.
Then we know if the camera setup was successful or failed.

lostgab, 2017-03-07:
ok, ill get it done and see what happens:)ty

syadav8126, 2017-04-18:
In file included from C:\Users\sb\Documents\Arduino\libraries\LiveOV7670Library/BufferedCameraOV7670_QQVGA_10hz.h:8:0,
                 from C:\Users\sb\AppData\Local\Temp\build8092868746636873910.tmp\ExampleTftBufferedCameraFrame.cpp:13:
C:\Users\sb\Documents\Arduino\libraries\LiveOV7670Library/BufferedCameraOV7670.h:46:17: error: 'constexpr' does not name a type
   inline static constexpr Tx getLineLength() __attribute__((always_inline));

Facing this error.....please suggest me some solution.
thank you.

indrekluuk, 2017-04-18:
What version of Arduino IDE are you using?

"constexpr" is c++11 feature.
c++11 should be enabled for Arduino IDE for some time now.

syadav8126, 2017-04-19:
i am using arduino ide version 1.6.5 ..
i replace" constexpr "to "cont" and resolved this problem..
the next error is "Resolution:: does not found a class or  namespace"....for that i just erase "Resolution"..and got resolved ..
now ide is asking to add some header file like ; "queue.h","twi.h","wire.h".
So does it have to be happen or I'm doing something wrong?
Thanks for your support

syadav8126, 2017-04-19:
please suggest me proper ide and full procedure...,like the changing "src" to "Live7670".
Thanks in advance. I am looking forward to your answer..

indrekluuk, 2017-04-19:
With the latest Arduino IDE it should compile without any errors. I think c++11 was supported from version 1.6.8, but I am not 100% sure. 1.8+ should definitely work.

If you download the latest version of the code form Github then renaming the folder is not necessary anymore. The LiveOV7670.ino is already inside folder named "LiveOV7670"

Before compiling you have to copy "LiveOV7670Library" and "Adafruit_GFX_Library" from ./src/lib/ to Arduino "libraries" folder (It should be in your home folder: ~/Documents/Arduino/libraries)
If you already have "Adafruit_GFX_Library" in your "libraries" folder then you don't have to copy that.

syadav8126, 2017-04-19:
Thanks for your help sir... are you on whatapp so that we can solve our
query further or we have to only contact via email. Thanks again for your
support.
I am from INDIA. Thanks

syadav8126, 2017-04-20:
After code compilation what to do next ?
java have been downloaded , the three txrx files have been moved to their specific folder. Now what to do next.

syadav8126, 2017-04-20:
The code has been compiled successfully. How to show output of OV7670 camera on the screen using java. Please suggest something.

indrekluuk, 2017-04-20:
Did you get both projects compiling?
1. LiveOV7670 - https://github.com/indrekluuk/LiveOV7670
2. Java project ArduinoImageCapture - https://github.com/indrekluuk/ArduinoImageCapture

syadav8126, 2017-04-22:
The first code is Compiled  and in second, i have placed the rxtx files to
their place and installed msvc 2008. Now where to capture snap and where
can i see them?

indrekluuk, 2017-04-22:
Does Visual Studio support java?
I personally use IntelliJ for Java. Community eddition is free.

Did you manage to compile and run the java application? A window should open where you can select com port and baudrate.

syadav8126, 2017-04-22:
i do not know whether visual studio support java .
how to use intelliJ?
no java window is opening. It is getting complicated please tell me some solutiion.

indrekluuk, 2017-04-23:
Install and run IntelliJ Community edition.

import project->Select folder "ArduinoImageCapture"->(Create project from existing source) Next->Next->Next->Next->Next-> (Select your Java version. For me it was "C:\Program Files\Java\jdk1.8.0_131") Next -> Finish

Now you have to add run configuration.
  
1. In the upper right corner click on the small arrow and then "Edit Configurations.."
[image]  

2. Click on the "+" and select "Application"
[image]

3. Name the configuration and select main java class
[image]

4. Now if you have copied these files to correct location in JAVA folder
Copy RXTXcomm.jar ---> <JAVA_HOME>/jre/lib/ext
Copy librxtxSerial.so ---> <JAVA_HOME>/jre/lib/i386/
Copy librxtxParallel.so ---> <JAVA_HOME>/jre/lib/i386/
You should be able to run the application by clicking the green arrow in the upper right corner.
[image]

And application window should open
[image]

leonhart88, 2017-07-01:
how is it works on arduino mega 2560?
i've already compile using Arduino Uno it's working but in Arduino/Genuino Mega or Mega 2560 board and Processor ATmega 2560(mega2560). But the message shows :

In file included from C:\Program Files (x86)\Arduino\libraries\LiveOV7670Library/BufferedCameraOV7670.h:8:0,

                 from C:\Program Files (x86)\Arduino\libraries\LiveOV7670Library/BufferedCameraOV7670_QQVGA_10hz.h:8,

                 from C:\Users\Anthony\AppData\Local\Temp\buildd92f8e36536f19f59eabeec9e696f075.tmp\sketch\ExampleTftBufferedCameraFrame.cpp:13:

C:\Program Files (x86)\Arduino\libraries\LiveOV7670Library/CameraOV7670.h: In member function 'void CameraOV7670::waitForVsync()':

C:\Program Files (x86)\Arduino\libraries\LiveOV7670Library/CameraOV7670.h:133:10: error: 'OV7670_VSYNC' was not declared in this scope

   while(!OV7670_VSYNC);

          ^

C:\Program Files (x86)\Arduino\libraries\LiveOV7670Library/CameraOV7670.h: In member function 'void CameraOV7670::waitForPixelClockLow()':

C:\Program Files (x86)\Arduino\libraries\LiveOV7670Library/CameraOV7670.h:142:9: error: 'OV7670_PIXEL_CLOCK' was not declared in this scope

   while(OV7670_PIXEL_CLOCK);

         ^

C:\Program Files (x86)\Arduino\libraries\LiveOV7670Library/CameraOV7670.h: In member function 'void CameraOV7670::waitForPixelClockHigh()':

C:\Program Files (x86)\Arduino\libraries\LiveOV7670Library/CameraOV7670.h:146:10: error: 'OV7670_PIXEL_CLOCK' was not declared in this scope

   while(!OV7670_PIXEL_CLOCK);

          ^

C:\Program Files (x86)\Arduino\libraries\LiveOV7670Library/CameraOV7670.h: In member function 'uint8_t CameraOV7670::readPixelByte()':

C:\Program Files (x86)\Arduino\libraries\LiveOV7670Library/CameraOV7670.h:150:10: error: 'OV7670_PIXEL_BYTE' was not declared in this scope

   return OV7670_PIXEL_BYTE;

          ^

Using library Adafruit_GFX_Library at version 1.1.5 in folder: C:\Program Files (x86)\Arduino\libraries\Adafruit_GFX_Library 
Using library SPI at version 1.0 in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI 
Using library LiveOV7670Library in folder: C:\Program Files (x86)\Arduino\libraries\LiveOV7670Library (legacy)
Using library Wire at version 1.0 in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\Wire 
exit status 1
Error compiling.

Help me Please...

indrekluuk, 2017-07-01:
I have modified the library so it will now compile for Arduino Mega (you have to re-download "LiveOV7670Library").

But I don't know if it actually works with Mega. I have not tested it.

leonhart88, 2017-07-01:
it's not working yet, but can be compiled
I've change the pin before for TFT LCD, because it's not working (TFT only). So, i change it to
SPI clock/SCL 	- PIN 13 (5V -> 3.3V)  --> change to 52
SPI data/SDA 	- PIN 11 (5V -> 3.3V) --> change to 51
should i change the connection pin on OV7670?

closed 2017-11-29 by indrekluuk