indrekluuk/LiveOV7670 #3: Shift level

anhtuanmanga, 2017-06-23:
DC - PIN 8 (5V -> 3.3V)
CS - PIN 9 (5V -> 3.3V)
RESET - PIN 10 (5V -> 3.3V)
SPI data - PIN 11 (5V -> 3.3V)
SPI clock - PIN 13 (5V -> 3.3V)
do i need connect these pins to shift level becasue i get white screen when use shift level
i get image when i don't use shift level when connect arduino and spi screen but my image not clear like your
can you help me?
sorry for my bad english

Rajeshlaguri, 2017-06-24:
Hi , The code by Indrekluuk for arduino worked perfectly for me . I used the same configurations i.e. 3.3V via level shifter .Make sure you are using correct pull up resistor for I2C clock and I2C data .Please refer to image for connections. 
If you are getting image without level shifter(but its better to use it as voltage rating of tft display is 2.8V to 3.4V ) then proceed with it . The OV7670 have one focus, so you have to adjust camera lens by rotating the lens .
And please recheck your connection as provided in software and your circuit connections. And make sure you are calling the correct function .
Wish you luck.

indrekluuk, 2017-06-24:
Like Rajeshlaguri said, the screen will work without level shifter, but it is not designed for that high voltage. For me the screen had faint horizontal stripes across the screen without a level shifter. Possibly the screen could die earlier if you use it with 5V.  
   
Camera clock (xclock pin) also needs a level shifter. I don't know if it can handle 5v.  
  
How does the image look exactly? Can you make a picture? Is it just blurry or are the colors messed up?

anhtuanmanga, 2017-06-25:
here i result i got
[image]

indrekluuk, 2017-06-25:
It seems like it is out of focus. Try to adjust the lens.

closed 2017-11-29 by indrekluuk