indrekluuk/LiveOV7670 #4: Image streaks (and darkness)

larsenglund, 2017-11-28:
Hi, thanks for this amazing piece of code! I spent a couple of hours trying to get the same thing working myself but only got like 0.5fps so I was very happy to find this (found it when giving up on atmega and was googling how to get it running faster on a STM32 :))

So it works fine except for some distorted lines / streaks in the image. First example is in greyscale mode:
[image]
And then in color mode:
[image]
This is what the camera is looking at in the color image:
[image]

Also, the image is overall very dark. Is if possible for the auto exposure to settle to a higher level? Or maybe turn it off altogether and setting it to a fixed value.

indrekluuk, 2017-11-29:
Hey!

Can you run the Adafruit test program (the one that draws text and all kinds of lines one the screen) just to make sure that the issue is not with connection to the screen? The issue probably isn't with the screen, but it would be good to make sure because it is easy enough to check.

It seems that either pixel reading is a little out of sync or one or more pixel wires have bad connection. How long are the wires from the camera to Arduino?

If you try a slow version (for example 2 FPS) does this still have the stripes?

larsenglund, 2017-11-29:
Hi! 
The Adafruit test program and the 2 FPS version of your code shows no stripes. The wires are about 20cm from the camera to the Arduino.
[image]

indrekluuk, 2017-11-29:
What does pixel clock signal look like from the Arduino's side if you measure it with oscilloscope? It should be PIN 12 if you connected it as shown in my schematic. On higher frequencies it may start distorting with longer cables and also the bread board connection in-between. I had everything mounted onto a PCB with very short connections.

Was the image brighter on lower FPS?

larsenglund, 2017-11-29:
I have connected it as shown in your schematic, with the exception of not using level shifters. I only use a voltage divider on XCLK and pullups for SIOC and SIOD:
[image]

The pixel clock looks like this (@10fps):
[image]

I think the image is equally dark, here is a comparison between 10 and 2 fps:
[image]

indrekluuk, 2017-11-29:
This looks OK.

Could you do the same measuring on the XCLK pin from the camera side. Maybe the voltage divider isn't fast enough. At 10FPS I am not checking the pixel clock PIN on each pixel because there is no time for that. I just detect the first pixel of a new frame and then start reading "blindly". If XCLK is uneven then it can get out of sync.

In lower frame rates code actually checks pixel clock pin.

larsenglund, 2017-11-29:
The XCLK on the camera side did not look good:
[image]

I change the voltage divider to a mosfet based bi-directional level shifter and got this:
[image]

The streaks disappeared (yay!) but now I get some kindof "animated/scrolling" purple/cyan errors in bright areas of the image and some barely visible stripes all over the image:
[image]

indrekluuk, 2017-11-29:
The new one actually seems worse than the voltage divider version. At least some of the bumps may count as clock cycle. It is probably because of the wire length.

I used the same kind of level shifter. For me it looks like this:
[image]

Maybe you can try voltage divider with lower resistor values? Arduino PIN can handle up to 40mA. Maybe 20mA to be safe. 5V/0.02A = 250 Ohm total (instead of 20k).

larsenglund, 2017-11-29:
That did the trick! Found a couple of 330 Ohm resistors first so tried with them and got this signal:
[image]

I can't thank you enough for helping me debug this! I hope it will help someone else in the future!
On a side-note, what FPS did you get with the STM32?

closed 2017-11-29 by larsenglund

indrekluuk, 2017-11-29:
You are welcome!

Not much higher than Arduino. Maximum reliable image I got was at 12FPS.

With Arduino I got the camera and Arduino to be in perfect sync so I could read pixel data while ignoring pixel clock.

For some reason I could not do that with STM32. 
There is a topic for OV7670 on http://www.stm32duino.com/ forums. I read that it is probably possible to capture pixels on hardware level with STM32. So I believe it is possible to go higher than 12FPS but I haven't tried it since my initial goal was to get it working with Arduino. I ported it to STM32 just to see if I can get it working.

Is the image still too dark? The camera object has methods "setContrast" and "setBrightness" you can try those. Although for me the default settings work pretty well.

larsenglund, 2017-11-29:
Ok, thanks for the STM32 info!

It's still a bit dark. I will try to fiddle with the histogram auto exposure thresholds.

I got an idea for improving framerate: since you are blindly reading pixels, could you not double the framerate for grayscale video by doubling the pixel clock, changing to YUV and only reading every other byte (the luma (Y)) from the YUYVYUYVYUYV.. byte stream?

larsenglund, 2017-11-29:
I ordered a couple of your PCB:s from ALLPCB by the way :) They offer free world wide DHL shipping so I only paid $5.5 for 10pcs, seems too good to be true, I guess time will tell :)

indrekluuk, 2017-11-30:
That does sound too good. Last time I ordered from PCBWay. They also have 5 USD for 10pcs (up to 10cm10cm size) but DHL delivery is around €25 (or China Post which is cheaper but takes over three weeks to arrive).

Next time I'll try ALLPCB. $5 isn't too much of a sacrifice if it isn't as promised.