indrekluuk/LiveOV7670 #6: Use with ESP8266?

larsenglund, 2017-12-02:
I was thinking if it would be possible to use the camera with a ESP8266. The idea is to use the fast SPI (40MHz) of the ESP to provide the XCLK and also to read pixel data from a shift register connected to the datalines from the camera (and using PCLK to latch data in the shift register). Do you think this is doable?

Running the ESP at 160MHz would hopefully give enough time to push out rows of the image as UDP packets turning it into the cheapest webcam ever :) Or give time to do some actual image processing on the data for use in robots. Maybe it's even possible to use some cleverness to be able to switch between reading pixels from the camera on the SPI to pushing them out to a display on the same SPI. Or maybe software SPI is enough at 160Mhz to get a good framerate on the display.

indrekluuk, 2017-12-03:
Reading from the camera is very time critical. Don't you have to share time with WIFI processes? 

Using FIFO version of OV7670 would solve all the problems here. It also has its own oscillator for clock on the PCB.
https://www.ebay.com/itm/640-x-480-VGA-CMOS-AL422-3M-Bits-OV7670-FIFO-Camera-STM32-Driver-Module-I2C/222657364015?hash=item33d769c02f:g:GaEAAOSwLdBaCpuA

indrekluuk, 2017-12-03:
Ok, I understand it from the perspective of challenge :)

The reason I made the Arduino version was also to prove to myself that it can be done. Specially because read from different places that Arduino is too weak for the camera.