indrekluuk/LiveOV7670 #9: Ethernet

kozuch, 2018-07-26:
Is it possible to add ethernet to the OV7670 solution (without fifo)? I would like to use the Waveshare LAN8720 PHY as advised here:

https://github.com/espressif/esp-idf/blob/master/examples/ethernet/ethernet/README.md
https://sautter.com/blog/ethernet-on-esp32-using-lan8720/

Thanks!

indrekluuk, 2018-07-27:
I think the hardest part of doing it is the fact that without buffering the wifi/ethernet module has to be able to stop everything what it is doing to receive the camera data. Maybe it can be done since wifi modules are much faster then Arduino, but I haven't tried it myself yet.  
  
At some point I want to try if I can get it working with the ESP8266 wifi module.

kozuch, 2018-07-28:
Thanks for your reply! I am Arduino noob. Are there enough gpio pins left on Arduino for Ethernet? I can sacrifice the Spi display as I do not need it.

indrekluuk, 2018-07-28:
It depends on the ethernet module. If the module can receive data over SPI then there probably is enough left-over PIN's to do that (if you remove the display).