indrekluuk/LiveOV7670 #10: Framerate

kozuch, 2018-07-28:
What is the framerate you achieved? It probably depends on the Arduino model used, right? Did you test FPS limits of various boards?

I need 320x240 at 15 fps with 8 bit pixel data (yuv etc.). Can you suggest a board for this please? I want sensor without fifo and also Ethernet capability. Can Arduino do this (maybe Due) or is something like STM32 or ESP32 needed?

Thanks in advance!

indrekluuk, 2018-07-28:
I have only tried with Arduino Nano/Uno and STM32 "blue pill" board. For Arduino 16Mhz clock rate sets the limit and there isn't any time left to squeeze more out of it. 

With the STM32 (72Mhz) I was reliably getting it to work 160x120@12fps and semi-reliably @15fps. But here with enough tweaking and twiddling it should be possible to get more out of it. I got the STM32 project into good enough state for me and haven't experimented it with after that.  

I haven't used Arduino Due myself but it seems to have similar specs to the STM32 "blue pill" board. So what you want (320x240 at 15 fps) might be possible but it requires lots of experimentation.

At some point I want to experiment with the esp8266 wifi module. It is much faster (400Mhz) but it has to share work load to do its wifi stuff. And also it doesn't have many IO ports. So maybe the solution is to add some shift register chips between camera and the module so that SPI can be used to get data to the wifi module.

kozuch, 2018-07-28:
Thanks for your reply! It is a real challenge to deal with image sensors definitely! :)

I saw the [bitluni's post](http://bitluni.net/esp32-i2s-camera-ov7670/) on ESP32 - he uses I2S for parallel data - not sure if this helps wih framerate etc. Also, I noticed that higher end STM32s do have a Digital camera interface (DCMI - [see here](https://www.st.com/content/ccc/resource/technical/document/application_note/group0/c0/ef/15/38/d1/d6/49/88/DM00373474/files/DM00373474.pdf/jcr:content/translations/en.DM00373474.pdf)) - if I understand things correctly this interface may offload the camera sensor data acquisition from the CPU so that higher framerates may be possible?

I would like to keep things as simple as possible and would like to explore all possibilities before stepping up to dedicated camera interfaces (DCMI).