indrekluuk/LiveOV7670 #12: Porting to Teensy

ArsenioDev, 2018-08-16:
Hey there, found this project and I've decided to start porting it over to the massively powerful Teensy 3.2. 
I've already got a 30*30 mouse sensor clocking out frames at well in excess of 45FPS bottlenecked by the display method. Plus it can be overclocked to up to 166mhz, though 144mhz (a 100% overclock) is rock solid. First issue is the i2c wire lib and complaints it has

indrekluuk, 2018-08-17:
That's cool! With that much higher clock rate you probably can get higher resolution also.
Can you actually get the camera to give out 45FPS? According to the documentation it works up to 30FPS.

ArsenioDev, 2018-08-18:
Ah, I was not referring to the OV7670, rather the ADNS-3080 for the 45fps on display. 
Still having issues getting the SCCB interface talking for the 7670