indrekluuk/LiveOV7670 #13: save image to SD?

zoomx, 2018-12-10:
How about to save a single image on a SD? I believe that Teensy 3.5 or 3.6 can do that (once ported as I see on Issue #12), but I wander if it is possible with an STM32.

indrekluuk, 2018-12-10:
I am not sure, maybe it can be done. stm32 (blue pill board) has 20kB of ram. It is not enough to hold a frame but it might be enough to to buffer the image while reading from camera->writing to SD card. The main problem is that writing to SD card is slow

josephresearcher, 2022-02-16:
https://learn.linksprite.com/jpeg-camera/use-linksprite-jpeg-camera-to-take-picture-and-store-to-a-sd-card/

josephresearcher, 2022-02-16:
please add this option :

https://www.linksprite.com/tutorial-of-using-jpeg-camera-to-capture-a-picture-and-save-to-sd/