indrekluuk/LiveOV7670 #17: How to make this code work for OV2640...?

sriranjanr, 2019-10-07:
Hello,

I have been trying to get this code to work with OV2640 but I am unable to. I always get red screen I connect the camera half way point and I am not getting a green screen. Please let me know what I should do?

indrekluuk, 2019-10-07:
Hello!

Since it is not the same camera the configuration is probably different (I don't know how much different).
You should try to find some code that is specifically designed for OV2640.

sriranjanr, 2019-10-09:
I was able to get it to a state to make it activate the green blank frame but after this I am not able to get any image from it. As many registers of 7670 and 2640 are different, I am unable to figure out the register configuration and other magic/mystery registers. Please guide me.

sriranjanr, 2019-10-09:
For once I got some garbled pixel data but the rest of the times it was only green frame. This 0V2640 has a HREF pin. So how do I use this pin to make the sensor work....