indrekluuk/LiveOV7670 #20: Help! BufferedCameraOV7670_QQVGA_10hz.h; No such file or directory

GijsLB, 2020-12-16:
I have an error when I verify the whole project. The ExampleTftBufferedCameraFrame.cpp file shows an error as such above. I am quiet new to Arduino, but love to learn. Can someone help? :) thanks in advance

indrekluuk, 2020-12-16:
copy "src/lib/LiveOV7670Library" and "src/lib/Adafruit_GFX_Library" to Arduino "libraries" folder (If you already have "Adafruit_GFX_Library" then you don't have to copy that)

GijsLB, 2020-12-16:
Hey! Both those libraries are installed; I did it manually. They also appear in Sketch > Include Library

indrekluuk, 2020-12-17:
This is strange. Did you restart Arduino IDE after installing the libraries?

Alternatively, you can try to copy all the content of the libraries into the sketch folder (next to the .ino file). You may be required to replace the #include <> with #include ""

GijsLB, 2020-12-17:
Yes, I restarted the Arduino IDE after installing the libraries. By copying all the library content into the sketch folder ( ctr + k) (right?) there seems to be no difference, also after restarting the file. I must be doing something wrong. See the image I've attatched. The right two folders are the libraries, the left folder is the sketch folder. Many thanks
[image]

Greetings from Holland

indrekluuk, 2020-12-17:
If you copy the content next to the .ino file then you should also change the <> to ""
For example 
#include <BufferedCameraOV7670_QQVGA_10hz.h>
to
#include "BufferedCameraOV7670_QQVGA_10hz.h"

But did you at first copy the two libraries into the "C:\Users\<your user name>\Documents\Arduino\libraries" folder? This is the correct location and it should work. It is strange that it didn't work for you.

N8delta, 2023-11-16:
I have the same problem

nkravitz99, 2026-03-14:
I'm having the same issue, has anyone been able to resolve it?