indrekluuk/LiveOV7670 #24: Energia CC1350 Version

alperkrt, 2021-02-01:
Hi,
Can you edit and give the code that works with Texas Instrument CC1350 and for Energia IDE. It is very similar with arduino. However, UDR0 is not working with CC1350. If you give a version for CC1350 (Energia). It would be amazing. Camera still is the same OV7670. I will use UART to see the real time video. Thank you for your time. 
sincerely

indrekluuk, 2021-02-02:
It would need lots of experimentation to get it working with another chip. Currently, I don't have the time and I also don't have the CC1350 chip.

alperkrt, 2021-02-03:
Thank you so much for your response. If you have a time, could you share the video of the code by explaning the code and algorithms. If we learn the algorithms behind the code we can convert this code into anything that we want. However, If you are too busy at this time, I would undersand that. again thank you so much for your vaulable response and time.

indrekluuk, 2021-02-06:
The most important files are (and their direct dependencies):
https://github.com/indrekluuk/LiveOV7670/blob/master/src/lib/LiveOV7670Library/CameraOV7670.h
https://github.com/indrekluuk/LiveOV7670/blob/master/src/lib/LiveOV7670Library/CameraOV7670.cpp

At the top of the CameraOV7670.h are platform-specific definitions. Which should all be defined for the new microcontroller.

It would be easiest to first get the i2c configuration connection working. After that, you can try if you can detect the pixel clock and vsync signals. If that also works then you can try to compose an image from the received data.