indrekluuk/LiveOV7670 #25: Led and button

DenisaAlice, 2021-04-16:
Congrats for your project, it helps me very much! I have a question: how can i use a button and a led in this project? 
I want to press a physical button on the A. uno, the led to become on and pixel reading to begin, but only for one frame until I push the button again.
I tried to modify ExempleUart.cpp and function processFrame() but i only managed to make the led light up when the pixel reading begin.
I'm talking about the project with the java application.
Please give me a helping hand in changing the code. Thank you !

indrekluuk, 2021-04-18:
Hey!

You can do something like this:

bool isFrameSent = false;
void processFrame() {
  if (isButtonPressed() && !isFrameSent) {

    ... frame sending code ...

    isFrameSent = true;
  } else {
    isFrameSent = false;
  }
}

where isButtonPressed() is a function that checks the button's pin.

Asinedecila, 2021-04-18:
> Hey!
> 
> You can do something like this:
> 
> ```
> bool isFrameSent = false;
> void processFrame() {
>   if (isButtonPressed() && !isFrameSent) {
> 
>     // frame sending code here
> 
>     isFrameSent = true;
>   } else {
>     isFrameSent = false;
>   }
> }
> ```
> 
> where isButtonPressed() is a function that checks the button's pin.

Thank you very much for your answer, I'll try  and come back with a feedback.

Asinedecila, 2021-04-20:
Hey ! I tried what you told me, but something is still wrong. This is the code that I changed:

**int buttonState=0;

bool isButtonPressed(){
  pinMode(12,INPUT);
  buttonState=digitalRead(12);
  if(buttonState==HIGH)
  {  digitalWrite(8,HIGH);
    }
    else { digitalWrite(8,LOW);
  
       }
  }
bool  isFrameSent=false;**

// this is called in Arduino loop() function
void processFrame() {
  
  startNewFrame(uartPixelFormat);

  noInterrupts();
  camera.waitForVsync();

  for (uint16_t y = 0; y < lineCount; y++) {

#if UART_MODE==7
    // separate loop for full VGA
    for (uint16_t x = 0; x < lineLength; x++) {
      // ignore first pixel byte
      camera.waitForPixelClockRisingEdge();

      // second byte is grayscale byte
      camera.waitForPixelClockRisingEdge();
      uint8_t pixelByte;
      camera.readPixelByte(pixelByte);
      sendPixelByteGrayscale(pixelByte);
    }
    // delay for last byte
    pixelSendingDelay();

#else

    lineBufferIndex = 0;
    uint8_t sendWhileReadCounter = 0;

    lineBuffer[0] = 0; // first byte from Camera is half a pixel
  //digitalWrite(8,LOW);
    for (uint16_t x = 1; x < lineLength*2+1; x++) {
      
  
      // start sending first bytes while reading pixels from camera
       if (uartSendWhileReadingCount > 0) {
       **if(isButtonPressed() && !isFrameSent){**   //I don't know if this is the line where I should put this if
        if (sendWhileReadCounter) {
          sendWhileReadCounter--;
          
        } else {
          sendNextPixelByte();
          sendWhileReadCounter = uartSendWhileReadingCount;
          
        }
      }
      camera.waitForPixelClockRisingEdge();
      camera.readPixelByte(lineBuffer[x]);

    // send rest of the line
    while (lineBufferIndex < lineLength * 2) {
      sendNextPixelByte();
      pixelSendingDelay();
    }
#endif

    endOfLine();
  
 **isFrameSent=true; //This is also what I wrote
} 
  else isFrameSent=false;
     }**
  interrupts();
  
}

  frameCounter++;
  debugPrint("Frame " + String(frameCounter));
  //debugPrint("Frame " + String(frameCounter, 16)); // send number in hexadecimal
  
}

I think there is also a conflict between the java application with the LISTEN button and the button that I put on the arduino because I got this: 
[image]
[image]

I need this code in one of my school projects, please help me solve this.

indrekluuk, 2021-04-20:
Hey!

The "pinMode(13,INPUT);" should go inside "void initializeScreenAndCamera()"

```
void initializeScreenAndCamera() {
  pinMode(13,INPUT);
  ...
```

Currently isButtonPressed() doesn't return a value. It should be something like this:
```
bool isButtonPressed() {
  buttonState=digitalRead(13);
  if(buttonState==HIGH) {
    digitalWrite(8,HIGH);
    return true;
  } else {
    digitalWrite(8,LOW);
    return false;
  }
}
```

The "if(isButtonPressed() && !isFrameSent){" should surround everything inside "processFrame()"

```
void processFrame() {
  if(isButtonPressed() && !isFrameSent){ // button check starts here

    startNewFrame(uartPixelFormat);

    noInterrupts();
    camera.waitForVsync();

    for (uint16_t y = 0; y < lineCount; y++) {

#if UART_MODE==7
      // separate loop for full VGA
      for (uint16_t x = 0; x < lineLength; x++) {
        // ignore first pixel byte
        camera.waitForPixelClockRisingEdge();

        // second byte is grayscale byte
        camera.waitForPixelClockRisingEdge();
        uint8_t pixelByte;
        camera.readPixelByte(pixelByte);
        sendPixelByteGrayscale(pixelByte);
      }
      // delay for last byte
      pixelSendingDelay();

#else

      lineBufferIndex = 0;
      uint8_t sendWhileReadCounter = 0;

      lineBuffer[0] = 0; // first byte from Camera is half a pixel

      for (uint16_t x = 1; x < lineLength*2+1; x++) {
        // start sending first bytes while reading pixels from camera
        if (uartSendWhileReadingCount > 0) {
          if (sendWhileReadCounter) {
            sendWhileReadCounter--;
          } else {
            sendNextPixelByte();
            sendWhileReadCounter = uartSendWhileReadingCount;
          }
        }

        camera.waitForPixelClockRisingEdge();
        camera.readPixelByte(lineBuffer[x]);
      }

      // send rest of the line
      while (lineBufferIndex < lineLength * 2) {
        sendNextPixelByte();
        pixelSendingDelay();
      }
#endif

      endOfLine();
    }
    interrupts();

    frameCounter++;
    debugPrint("Frame " + String(frameCounter));
    //debugPrint("Frame " + String(frameCounter, 16)); // send number in hexadecimal

  
    isFrameSent = true;  // button check ends here
  } else {
    isFrameSent = false;
  }
}
```

Please note that I have not tested it myself, but it should work with code something like this.

Asinedecila, 2021-04-20:
Hey again ! IT S WORKINGGGG !!! Thank you very very much! YOU'RE AWESOME!

indrekluuk, 2021-04-21:
Hey! That's great!

closed 2021-04-21 by indrekluuk