indrekluuk/LiveOV7670 #27: Problem with installing "ArduImageCapture" Plug-In

ArmanPixel82, 2021-11-23:
I copy the plugin in to the correct address but it doesn't appear in tools bar.
[image]
.

indrekluuk, 2021-11-23:
Hey!

If you go inside the "ArduImageCapture" folder does it look like this:
[image]
(the .properties file may be missing. That is OK)

Is the sketchbook location set to "Documents/Arduino" in Arduino IDE?
[image]

ArmanPixel82, 2021-11-24:
The sketchbook location was wrong :))
Thanks for your help

Gururam1978gs, 2022-01-14:
This step is not working, though the captured the folder correctly and matching the sketchbook location

augustvanhout, 2022-11-12:
Be sure to use legacy IDE until they implement custom tools! Use <2.0 and it may fix the issue.

mikcip, 2023-08-18:
Nekem nem jelenik meg az "Eszközök" listáján az ArduImageCapture, pedig mindent leellenőriztem, és rendben találtam. Örülnék, ha valaki tudna segíteni.