indrekluuk/LiveOV7670 #28: How can I make VSYNC pulse when I need to capture frame?

pome83, 2021-12-30:
So topic is the problem. I have movement sensor and I want to make camera starting the frame by interrupt from the sensor to get image immediately when moving sensor is triggered. So how can I reset that VSYNC timer from the program?

ElianVasquez, 2022-06-27:
Hey, have you created a way to do that? I need it asap

ElianVasquez, 2022-06-27:
> So topic is the problem. I have movement sensor and I want to make camera starting the frame by interrupt from the sensor to get image immediately when moving sensor is triggered. So how can I reset that VSYNC timer from the program?

Hey, have you created a way to do that? I need it asap

ElianVasquez, 2022-06-27:
@pome83 Hey, have you created a way to do that? I need it asap

ElianVasquez, 2022-06-27:
> So topic is the problem. I have movement sensor and I want to make camera starting the frame by interrupt from the sensor to get image immediately when moving sensor is triggered. So how can I reset that VSYNC timer from the program?

@pome83 @indrekluuk Hey, have you created a way to do that? I need it asap