indrekluuk/LiveOV7670 #40: green square and no Vsync no matter what I do

DeDBoGDaN, 2023-09-07:
I have a green square all the time, everything is connected correctly and reliably, but no matter what I do, I always get a green square. But what I noticed is that on pin 3 of the Arduino Uno (Chinese replica) I have 2.5V instead of 5V, so maybe there is some power missing. If you download any other code, all digital pins work correctly and produce about 5 volts. So the problem is probably in the code. I don't know how to solve it. 
P. S. If I remove and reinsert the VSYNC port connector or briefly apply 3.3V to pin D3 (before the 680 ohm resistor), I get a Vsync notification, but not the camera image. I also tried to connect this pin without a voltage divider, but the result is the same: a green square without vsync notification and a maximum voltage of 2.5V. Even if nothing is connected, and only this code is loaded, the pin has only 2.5V.

indrekluuk, 2023-09-09:
Hey!

If you measure PIN 3 with a multimeter then 2.5V is normal. It is the clock signal for the camera which is a square wave altering quickly between 0 and 5V. Multimeter sees the average of the wave 2.5V. You should measure it with an oscilloscope if you want to see the actual signal.

PIN 3 with the voltage divider is XCLK not VSYNC. VSYNC should go to PIN 2. 
So if you manually pull PIN 2 to 0V and then apply 3.3V or 5V on it (which creates a rising edge signal) then you should also see the VSYNC message.

The VSYNC signal should appear periodically once per camera frame if all the wires are connected correctly:
https://www.youtube.com/watch?v=R94WZH8XAvM&t=434s
If you can't see the VSYNC message then there probalby is something wrong with the VSYNC connection.

DeDBoGDaN, 2023-09-09:
Thank you very much for your answer!
Here's what I'm doing (everything is connected exactly right): I'm working in UART 5 and 16 modes (doesn't work with other colours and black and white), press Listen, wait until the green square is filled, wait another second, connect Vsync to a 3.3V source (doesn't work with 5V), get a Vsync notification, start downloading the frame, get about half a frame, when more when less, and then it's dark. The images are attached. The following frames are also dark.
UART 5:
[image]
UART16:
[image]
and also when testing all UART mods, connecting Vsync to 3.3V or 5V stopped working. Most likely I burned it in some way

DeDBoGDaN, 2023-09-09:
also now I having like 50% chance to get green or red square) OH while i was typing this it's starts working but REALLY SLOOOOW in UART mode 5 (black square)

closed 2023-09-12 by DeDBoGDaN